PaxInstruments/t400-electronics #118: Power on/off output low

charlespax, 2015-01-04:
Power on/off goes output low and turns the system off all the time.

(From our discussion at lunch.)

samchoy88, 2015-01-05:
As the PWR_ONOFF pin set as output Low initially, it cause the systemturn off after JP1 has been shorted.
[image]

I may change circuit design to prevent this issue.

charlespax, 2015-01-05:
Background: the purpose of having JP1 in this design is to aid development by having an easy way to keep the device always using hardware rather than software.

If the jumper is not shorted and the power switched is pressed, the system will boot and set PWR_ONOFF to HIGH. Then the system will stay on until PWR_ONOFF is set to LOW.

If the jumper is shorted and PWR_ONOFF is set to LOW, logic input A will be LOW. If there were no 1k resistor in series with JP1, there would be a short, which is bad.

If the jumper is shorted and PWR_ONOFF is set to HIGH, logic input A is always HIGH. No problems in this case.

I think we need a jumper that can short logic input A to VBUS and a resistor between PWR_ONOFF and logic input A. In this case the Jumper can be shorted to VBUS and always keep logic input A HIGH. PWR_ONOFF can be high or low and not be shorted to GND because there would be a resistor.

samchoy88, 2015-01-16:
#134

charlespax, 2015-01-16:
How was it modified? Screenshot please.

samchoy88, 2015-01-16:
Changed logic gate to NAND gate. All switch and jumper pin will trigger the circuit by Low level.

Before:
[image]

After:
[image]

charlespax, 2015-01-16:
Beautiful :-)

MLXXXp, 2015-01-16:
I thought you were going to increase the value of the pullups from 10K to 100K, as per comments in issue #134.

samchoy88, 2015-01-16:
You're correct.
[image]

charlespax, 2015-01-16:
Done. Closing issue.

closed 2015-01-16 by charlespax