PaxInstruments/t400-electronics #120: Resistors are close together

charlespax, 2015-01-04:
There are some resistors that are a bit close together. Switching to 0603 should solve this.

(From our discussion at lunch.)

samchoy88, 2015-01-05:
Yes, I found the resistor are too close at the area of 3.3Reg part. It made a little bit difficult to solder by hand.
[image]

charlespax, 2015-01-05:
Let's increase the distance in Electronics version 0.10.

samchoy88, 2015-01-07:
Switched to 0603 resistor and capacitor does not improve much on space.
They looks almost compatible to each other.
Capacitor in Yellow is 0805, Green is 0603

It seem I need to find a better place and space for these circuit.

[image]

samchoy88, 2015-01-25:
More space between resistors in Rev 0.10
[image]

charlespax, 2015-01-25:
Closing issue.

closed 2015-01-25 by charlespax