PaxInstruments/t400-electronics #122: Longer soler pads on 2x3 header footprint

charlespax, 2015-01-04:
If we use the SIM card connector, we may want to make the 2x3 header footprint longer, so they do not have to be aligned perfectly.

https://github.com/PaxInstruments/t400-electronics/issues/121

samchoy88, 2015-01-05:
the programming port will apply larger pad that fit with SIM card connector:
[image]

[image]

[image]

charlespax, 2015-01-05:
Cool. Let's get this in Electronics version 0.10.

samchoy88, 2015-01-07:
Library changed but not significant.

Before:
[image]

After:
[image]

samchoy88, 2015-01-22:
Due to limitation of the space, we give up the longer pads on 2x3 header footprint.
Changed back to normal size.

charlespax, 2015-01-22:
Moving issue to Electronics version :-)

charlespax, 2015-08-08:
Will not do. Closing issue.

closed 2015-08-08 by charlespax