PaxInstruments/t400-electronics #123: Inductor between isothermal plane and GND

charlespax, 2015-01-04:
Maybe we should use an inductor between the isothermal plane and ground.

(From our discussion at lunch.)

samchoy88, 2015-01-05:
Currently the Analog ground and Digital ground are connected by Ferrite Bead which can supresses high frequency noise pass though it. Hope it helps.

charlespax, 2015-01-05:
The most recent version in git still has a 10k resistor. The prototype board you made have the ferrite bead instead?

[image]

samchoy88, 2015-01-05:
Yes, the prototype board I made have the Ferrite bead instead. Let see the result.

charlespax, 2015-01-05:
Let's make the change for Electronics version 0.10. If it doesn't work out, we can revert.

charlespax, 2015-01-12:
From our conversation we have decided to use a zero ohm resistor. We should make a component for the library that is a resistor pad set with a trace linking them.

samchoy88, 2015-01-15:
I found the ground should use zero resistor instead.
However, the isothermal ground (Analog ground) and Normal ground should connect near power supply.
However, I cant find a good position for the power supply to achieve this. Any suggestion on this?
[image]

charlespax, 2015-01-15:
Whether the device is powered via USB or battery all the power goes through the 3.3 V regulator circled in the image below. Maybe I don't fully understand the concept, but I don't think the charging portion has much to do with this.

[image]

Can we run a trace from the isothermal ground to near the 3.3 V regulator?

samchoy88, 2015-01-15:
Here is the schematic for reference
[image]

As you can see the system can power by 2 different source, USB and Battery.
- You will find the current pass through the bottom part to charge battery and battery past through the bottom part back to regulator and supply the whole circuit by 3.3V.
- Besides, the analog GND must past through the noise digital GND to get back power supply (orange path). This made the Analog part noise and it may affect accuracy of the reading.

[image]

here is my suggestion:
- Move the all noisy digital component (SD card and Flash) to left hand side.
- RTC and its battery move to right hand side and made more space for regulator, charger

charlespax, 2015-01-15:
I see no problem with moving the SD card and flash to the left side. However, we need feedback from @starno. @starno what do you think?

What will happen with the BL-5C batter connector? Will that get moved to the left side too? That would require significant work on the enclosure, but is doable. @starno any thoughts on this part?

samchoy88, 2015-01-16:
Why BL-5C connector move together with SD-card connector?

charlespax, 2015-01-16:
It doesn't have to. It jus sounded like you wanted to move the battery connector too because it is part of the power system on the right side.

samchoy88, 2015-01-16:
Yes, the battery connector at bottom right side is OK. As the circuit density is low and it seem great for Analog GND and Digital GND connect together.

charlespax, 2015-01-25:
These two issues are not pretty much the same https://github.com/PaxInstruments/t400-electronics/issues/59

samchoy88, 2015-01-25:
Power distribution
[image]

samchoy88, 2015-01-25:
Isothermal plane connect to Normal GND by a short wire.

[image]
[image]

charlespax, 2015-01-25:
Cool. Closing issue.

closed 2015-01-25 by charlespax