PaxInstruments/t400-electronics #126: Panel design

charlespax, 2015-01-04:
We should make a panel design. We will need the PCBA house requirements.

This is a design we should have in-house and not something internal to the PCBA house.

(From our discussion at lunch.)

samchoy88, 2015-01-05:
Understand we want the panel design are the same after we switched to another PCB factory.
Stick with the plan like this?
If there is any comment, please feel free to let me know.
[image]

samchoy88, 2015-01-05:
Here are the information for panel design.
http://blogs.mentor.com/tom-hausherr/blog/2011/06/23/pcb-design-perfection-starts-in-the-cad-library-part-19/

charlespax, 2015-01-23:
This is for production. Moving to Electronics version 1.0 (production).