PaxInstruments/t400-electronics #129: Test pad size

charlespax, 2015-01-04:
Maybe we should have larger test pads to aid in programming jig alignment.

(From our discussion at lunch.)

samchoy88, 2015-01-05:
the programming port will apply larger pad that fit with SIM card connector:
[image]

[image]

[image]

charlespax, 2015-01-23:
@samchoy88 what are your thoughts on this? We should have a decision for Electronics version 0.10.

samchoy88, 2015-01-25:
Here is the drill size of the test pad.
[image]

This is the reference test point component dimension:
[image]

http://www.keyelco.com/category.cfm/Test-Points-color-keyed-Thru-Hole-Mount/Compact-Style/p/518/id/521/c_id/746

charlespax, 2015-01-25:
We only need a through hole test hook for a ground test point. The rest only need to be large enough to make contact with the test jig.

samchoy88, 2015-01-25:
Here is the sample test pin on the Rev 0.9.
[image]

Mounting holes at the corner will handle all test point alignment.

samchoy88, 2015-01-25:
Pogo pin work great on test point of Rev 0.9 board.
The pogo pin is sharp and narrow then the other (round head)
 
[image]

samchoy88, 2015-01-25:
Pogo pin reference:
http://www.pccp.com.tw/images/ccp/Tradition_Probes/P-25.pdf

charlespax, 2015-01-25:
Okay, closing issue.

closed 2015-01-25 by charlespax