PaxInstruments/t400-electronics #130: Electronics version 0.9 stopped working

charlespax, 2015-01-06:
I have two Electronics version 0.9 assembled boards from @samchoy88. I've been flashing one with firmware as I develop the T400 testing software. Now the board is not recognized by OSX. When I plug it in I get the following.

```
USBF:    1613.173    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 1 of Hub at 0x14000000)
USBF:    1613.474    The IOUSBFamily was not able to enumerate a device.
USBF:    1614. 30    The IOUSBFamily is having trouble enumerating a USB device that has been plugged in.  It will keep retrying.  (Port 1 of Hub at 0x14000000)
USBF:    1614. 30    The IOUSBFamily gave up enumerating a USB device after 10 retries.  (Port 1 of Hub at 0x14000000)
USBF:    1614. 30    The IOUSBFamily was not able to enumerate a device.
```

samchoy88, 2015-01-06:
Can you re-program the MCU by ICSP port with LilyPad Arduino USB bootloader?

charlespax, 2015-01-07:
I cannot reprogram the bootloader with any of my programmers on my OSX machine or on another Windows machine.

On one of the boards I removed the 32u4 and soldered on a new one. I successfully flashed the bootloader via ICSP.

I think there is a problem with the power being supplied from the wall. I can sometime feel a buzz on my computer. From now on I will not plug the T400 into my computer while it is plugged into the wall.

samchoy88, 2015-01-07:
It seem the computer is not grounded properly.
http://apple.stackexchange.com/questions/10545/is-it-bad-that-my-macbook-air-is-passing-me-electricity

charlespax, 2015-01-08:
Running on battery power I just killed another board! It seemed to happen after I accidentally released the SW_PWR button while the firmware was uploading.

A note from debugging yesterday: On a working board D+ and D- are not connected. On bad boards D+ and D- had a measurable resistance between them. When I get home I'll check the resistance between D+ and D- on the board I just killed.

charlespax, 2015-01-23:
I think this whole thing was a ICSP programmer issue. Closing.

closed 2015-01-23 by charlespax