PaxInstruments/t400-electronics #134: New soft power circuit

samchoy88, 2015-01-09:
I would like to implement the following soft power circuit and solve below issue:
- JP1 made circuit operate at 3.6V instead of 3.3V #117
- SW_PWR cannot detect button action #116

[image]

I choose NAND gate instead of OR gate and made all input pulled up by 10K resistor.
[image]

charlespax, 2015-01-09:
My only concern is pullup resistor on PWR_ONOFF is drawing 0.33 mA while the device is just sitting there.

Here's what the power on sequence looks.
```
A   B   Y
H   H   L   <-- SW_PWR is not pressed, device is off. Device if off.
H   L   H   <-- SW_PWR is pressed, device is off. User presses and holds the power button.
L   L   H   <-- SW_PWR is pressed, device is on. Device boots up and enables power. User is still holding power button.
L   H   H   <-- SW_PWR is not pressed, device is on. User releases power button.
```

samchoy88, 2015-01-09:
I think it can be 100K for pull up, as it only connect a logic which has high input impedance.

charlespax, 2015-01-09:
Cool. Let's make that change in Electronics version 0.10.

charlespax, 2015-01-16:
Done. Closing issue.
[image]

closed 2015-01-16 by charlespax