PaxInstruments/t400-electronics #136: Solar charging

charlespax, 2015-01-15:
A Kickstarter community member suggests "Can you add some solar cells and use rechargeable batteries to allow long duration measurements and not having to worry about energy source."

charlespax, 2015-01-15:
Solar charging would be cool. I think it would be best to have a system where the screen and device and be orientated for best usage while the solar panel can be placed in direct sunlight.

I think a solar powered battery pack connected to the microUSB port would be perfect for this.

[image]

Let's circle back to this idea in a future product.

closed 2015-01-15 by charlespax

Tsillen, 2015-01-15:
The next product will probably have a more energy efficient design. Thus depending on the mode it is in it can increase battery life dramatically (several months to even a year of battery life).

charlespax, 2015-01-15:
Especially if it doesn't have an LCD. Then we can integrate it all in one enclosure.

Tsillen, 2015-01-15:
Even with a display. As you can electronically disconnect it completely with an mosfet. Thus making the power draw of the lcd pretty much zero.