PaxInstruments/t400-electronics #137: LCD connector 1.0 mm footrpint

charlespax, 2015-01-16:
We need a footprint for an appropriate LCD connector for the LCD with a 1.0 mm pitch cable. The one in the image below should be good, but we need it to be 12 pins.

[image]

samchoy88, 2015-01-19:
Original LCD it has 12 pins only.
However you are replacing the 16 pins 1mm Pitch connector.
Please kindly advice the signal assignment on new 16 pins 1mm Pitch connector connector.

[image]

charlespax, 2015-01-19:
The connector is 12 pins. The 16 pin connector is a good template from which you can make a new connector.

samchoy88, 2015-01-19:
OK, I follow this part number dimension for reference
P/N: FH12-12S-1SH

[image]

charlespax, 2015-01-23:
We have four samples of the new LCDs, so we'll have them for Electronics version 0.10. Moving to the Electronics version 0.10 milestone.

samchoy88, 2015-01-25:
Both of the 1mm and 0.5 mm connector are in the PCB now.
See #140

charlespax, 2015-01-25:
It's in Github. Closing issue.
[image]

charlespax, 2015-01-25:
We should remove the 0.5 mm footprint when we have the custom LCD production run. See https://github.com/PaxInstruments/t400-electronics/issues/141.

closed 2015-01-25 by charlespax

charlespax, 2016-05-10:
See also https://github.com/PaxInstruments/t400-electronics/issues/162