PaxInstruments/t400-electronics #138: Adjust the thermal-couple connector width

samchoy88, 2015-01-16:
It seem the thermal couple connector is a little bit lose, 

[image]

I try to reduce the distance between connector (+) and (-), around 11mm ~ 12mm.

[image]

samchoy88, 2015-01-16:
See this improvement in Library
[image]

charlespax, 2015-01-16:
For the male mini thermocouple connector we have (8 mm center-to-center) + (2.4 mm/2) + (3.17 mm/2) = 10.785 mm edge-to-edge. That is at least 1.11 mm of play side-to-side. And if we look at the Keystone connector there appears to already be enough extra room. @samchoy88 how do you feel about reducing that 12 mm to 10.75 or 11 mm?

[image]
[image]

And when you change the footprint make sure the centerline of the foot print meets the centerline of the male thermocouple connector. The blades are different widths, so there will be an offset.

samchoy88, 2015-01-19:
As the thermal-couple has different size of terminal. (Positive pole is smaller then Negative pole).  Do you think we should apply the same size connector? (To avoid user plug in connector inversely)

[image]

[image]

Positive terminal: (3549 - For 0.110 tab) 
http://www.keyelco.com/product.cfm/Horizontal-Entry/Horizontal-Entry-Mini-Auto-Blade-Fuse-Clip/p/398/id/399/c_id/546/product_id/303

Negative terminal: (3572 - For 0.187 tab)
http://www.keyelco.com/product.cfm/Horizontal-Entry/3572/p/398/id/399/c_id/604/product_id/683

charlespax, 2015-01-19:
No have will come from plugging in backwards. The top panel is keyed, so a user cant do it. However, there is no harm in using two different width connectors as long as they work. The small connectors I've tried tend to deform. If you have connectors that work, I'm open to trying them in the next board revision.

samchoy88, 2015-01-19:
OK, I only apply the terminal: (3572 - For 0.187 tab)
http://www.keyelco.com/product.cfm/Horizontal-Entry/3572/p/398/id/399/c_id/604/product_id/683

[image]

However, the connector width is 5.64mm, the minimum side to side (2 terminals) width is 11.4mm (connector touch together). I made the connector side to side 12.2mm, reserved 0.8mm margin for extension while tab inserted. 

[image]

charlespax, 2015-01-25:
Looks good. Closing issue.
[image]

samchoy88, 2015-01-26:
I just print it out and verify the board again.
Do you think the connector are too close together?
[image]

charlespax, 2015-01-26:
The connectors in Electronics version 0.10 will be fine as long as we can make a gap. We can put Kapton tape to ensure there is no electrical connection.

In Electronics version 0.11 we are moving to a custom connector (https://github.com/PaxInstruments/t400-electronics/issues/17), so this is not an issue we will face in production.

closed 2015-01-26 by charlespax