PaxInstruments/t400-electronics #139: PCB mounting holes

charlespax, 2015-01-22:
The PCB standoffs on the enclosure now have a shoulder to secure the PCB more firmly (https://github.com/PaxInstruments/t400-enclosure/issues/64). This requires enlarging the PCB mounting holes. The keepout area on the PCB is the same, so this should not require any change to the layout or routing.

charlespax, 2015-01-22:
@starno What are the new dimensions of the PCB mounting hole?

starno, 2015-01-22:
4.5mm diameter in the same location

[image]

samchoy88, 2015-01-25:
Current mounting holes are 5.08mm diameter.
Is that OK?

[image]

[image]

charlespax, 2015-01-25:
That screenshot shows a 2.54 mm hole. I don't understand.

samchoy88, 2015-01-25:
Sorry, I made misunderstand on the value.
here is the corrected drill size
[image]

charlespax, 2015-01-25:
Looks good. When we manufacture the PCB we'll have to make sure the holes are not plated, which would reduce their diameter. Closing issue.

closed 2015-01-25 by charlespax