PaxInstruments/t400-electronics #140: Two LCD connector footprints

charlespax, 2015-01-23:
**NOTE: This is a nice-to-have feature, not a requirement. This feature is purely to aid testing and should not in any way compromise the production design.**

@samchoy88, if possible, it would be good to include both LCD connector footprints on the Electronics version 0.10 PCB. We only have four samples of the 1.0 mm pin pitch LCD, which would limit us to four active units for Electronics version 0.10. By including the second footprint we can have as many units as we like.

This is a quick sample of what the PCB could look like. It would be best to not have the connector hanging off the board, but can if that's what it takes to get the footprint on there.
[image]

charlespax, 2015-01-23:
If this doesn't work out for Electronics version 0.10, I can just roll an extended PCB that has the connector. This would be just for software development.

samchoy88, 2015-01-25:
Done
[image]

charlespax, 2015-01-25:
Looks good. Closing issue. We should remove the 0.5 mm footprint when we have the custom LCD production run. See https://github.com/PaxInstruments/t400-electronics/issues/141.
[image]

closed 2015-01-25 by charlespax