PaxInstruments/t400-electronics #141: Remove unused 0.5 mm LCD connector

charlespax, 2015-01-25:
In Electronics version 0.10 we kept the 0.5 mm and 1.0 mm LCD connector footprints for the transition from stock to custom LCD connector. Once we have the full  production run we can remove the 0.5 mm footprint.

See https://github.com/PaxInstruments/t400-electronics/issues/137

samchoy88, 2015-03-14:
removed unused 0.5 mm LCD connector in PCB layout. and remaining 1mm LCD connector footprint.
[image]

charlespax, 2015-03-14:
Perfect. I just got word that the new LCDs are in. We'll have plenty to work with.

charlespax, 2015-03-18:
The LCD factory is shipping the LCDs today.

Looks done to me. Closing issue.

closed 2015-03-18 by charlespax