PaxInstruments/t400-electronics #142: Review Arduino pinmap on silkscreen

charlespax, 2015-01-25:
Make sure the Arduino pinmap on the silkscreen is accurate. This should be the last issue addressed before ordering PCBs.

samchoy88, 2015-01-25:
Here is the silkscreen and schematic on Rev 0.10:

Pin map silkscreen position:
[image]

Pin Map on silk screen:
[image]

Pin map on Schematic: 
[image]

charlespax, 2015-01-25:
Here is the Arduino Leonardo pinmap. Does everything match? Is the silkscreen accurate?
[image]

samchoy88, 2015-01-26:
Are you sure the above pin mapping suitable for our project?
As I found T400 dos not connect UVcc to +5V and datasheet of ATmega32u4 also suggest not connect +5V to UVcc while I/O using 3.0V to 3.6V.

Reference: (Page 256) http://www.atmel.com/images/doc7766.pdf
[image]

charlespax, 2015-01-26:
The image I reference above is for the Arduino Leonardo. It is only used to correlate between physical pins on the ATmega32U4 and their Arduino digital/analog designations.

Don't worry about it, I'll take care of this issue.

charlespax, 2015-01-26:
ATmega32U4 | Arduino
---|---
1 | D7
2 | 
3 | 
4 | 
5 | 
6 | 
7 | 
8 | RXLED
9 | SCK
10 | MOSI
11 | MISO
12 | D11
13 | RESET
14 | 
15 | 
16 | XTAL2
17 | XTAL1
18 | D3, SCL
19 | D2, SDA
20 | D0
21 | D1
22 | TXLED
23 | 
24 | 
25 | D4
26 | D12
27 | D6
28 | D8
29 | D9
30 | D10
31 | D5
32 | D13
33 | 
34 | 
35 | 
36 | A0
37 | A1
38 | A2
39 | A3
40 | A4
41 | A5
42 | 
43 | 
44 |

samchoy88, 2015-01-26:
Yes, the Digital and Analog pin mapping are match with T400

charlespax, 2015-01-26:
Here is the pinmap for the current version of the T400

Arduino | ATmega32U4 | T400
---|---|---
D0 | 20 | SW_B
D1 | 21 | SW_A
D2 | 19 | SDA
D3 | 18 | SCL
D4 | 25 | LCD_A0
D5 | 31 | BATT_STAT
D6 | 27 | LCD_BL
D7 | 1 | RTC_INT
D8 | 28 | LCD_CS
D9 | 29 | VBAT_SENSE
D10 | 30 | VBAT_EN
D11 | 12 | FLASH_CS
D12 | 26 | LCD_RST
D13 | 32 | SW_C
D14 | 11 | MISO
D15 | 9 | SCK
D16 | 10 | MOSI
D17 | 8 | SD_CS
A0, D18 | 36 | DATA2
A1, D19 | 37 | DATA1
A2, D20 | 38 | SW_D
A3, D21 | 39 | SW_E
A4, D22 | 40 | SW_PWR
A5, D23 | 41 | PWR_ONOFF

[image]
[image]

charlespax, 2015-01-26:
All good. Closing issue.

closed 2015-01-26 by charlespax