PaxInstruments/t400-electronics #143: Connect buttons to interrupt pins

charlespax, 2015-01-26:
We should connect each button and RTC_INT to an interrupt pin. This will give us the highest flexibility if we have to change the UI later.

In Electronics version 0.10 the interrupts pins available for swapping are __PCINT4__ on LCD_CS, __PCINT5__ on VBAT_SENSE, __PCINT6__ on VBAT_EN, __PCINT7__ on FLASH_CS, and __PCINT0__ on SD_CS.

Item | Function | Interrupt | Note
---|---|---|---
__RTC_INT__ | Wake for readings | __INT6__ | Requires interrupt
__SW_A__ | Logging start/stop | __INT3__ | Requires interrupt
__SW_B__ | Logging interval | __INT2__ | Probably does not need interrupt
__SW_C__ | Temperature units | none | Should probably have interrupt
__SW_D__ | Toggle channels | none | Should probably have interrupt
__SW_E__ | Backlight | none | Should probably have interrupt
__SW_PWR__ | Power | none | Should probably have interrupt

For a pin map of Electronics version 0.10 see https://github.com/PaxInstruments/t400-electronics/issues/114.

cibomahto, 2015-01-26:
The only place where it really matters if the pin can generate an input is when the processor is in sleep mode. When the device is running, it makes more sense to use a single timer interrupt to poll all of the buttons, to make the debounce routine more effective.

From the manual, here is the list of interrupt sources that can be used to wake the processor:

[image]

As noted above, RTC_INT, SW_A, and SW_B are all connected to dedicated hardware interrupts (INTx) as well as pin change interrupts (PCINTx), so they are fine, and the other pins cannot be used to wake the processor. They happen to overlap for the Atmega32u4, but the two interrupt methods are separate peripherals- the pin change one will make an interrupt whenever the pin changes state (high-to-low and low-to-high), while the hardware interrupt can trigger on a low-to-high, high-to-low, or whenever the value is low. Either peripheral will work for waking up the t400.

The other buttons, also as noted above, aren't connected to an interrupt source and could not be used to wake the device from sleep.

cibomahto, 2015-01-26:
Here are the pins with external interrupt capability:

PB0: PCINT0
PB1: PCINT1
PB2: PCINT2
PB3: PCINT3
PB4: PCINT4
PB5: PCINT5
PB6: PCINT6
PB7: PCINT7
PD0: INT0
PD1: INT1
PD2: INT2
PD3: INT3
PE6: INT6

From a software perspective, it's cumbersome to have to deal with both types at the same time, but it's not a showstopper.

cibomahto, 2015-01-26:
Also note that the only interrupt that's capable of waking the system from sleep in the lowest power mode (all clocks off) is INT6, so placing the RTC interrupt on that line was a good call.

charlespax, 2015-01-26:
The table indicates "INT6,INT3:0 and pin change" can wake the processor. For example the ADC can only wake the processor from Idle and ADCNRM. Am I reading the table wrong?

cibomahto, 2015-01-27:
You are reading it correctly. Some of the peripherals are disabled in some of the sleep modes, so they can't be used to turn on the processor.

Also, detecting pin change interrupts while in the power down mode will slightly increase power usage, as a clock has to be enabled for the pin change peripheral to work. The INT6 pin's 'low' interrupt can work without a clock. Those details shouldn't matter with this design, because the difference should be small compared to the small amount that the rest of the peripherals use.

charlespax, 2015-01-29:
It looks like we'll never use the lowest level interrupt for RTC_INT. We would need at least one button on the same level, so a user can stop logging.

charlespax, 2015-02-04:
It looks like we can get all the buttons and the RTC_INT pin on interrupts. Let's do this for the next version.

charlespax, 2015-03-18:
Did this make it into the most recent board revision?

samchoy88, 2015-04-18:
Here are the pins with external interrupt capability:

PB0: PCINT0
PB1: PCINT1
PB2: PCINT2
PB3: PCINT3
PB4: PCINT4
PB5: PCINT5
PB6: PCINT6
PB7: PCINT7
PD0: INT0
PD1: INT1
PD2: INT2
PD3: INT3
PE6: INT6

Below is the pin assignment of the layout in Rev 0.12

Item | Function | Interrupt | Note
---|---|---|---
__RTC_INT__ | Wake for readings | __INT6__ | PE6
__SW_A__ | Logging start/stop | __INT3__ | PD3
__SW_B__ | Logging interval | __INT2__ | PD2
__SW_C__ | Temperature units | __PCINT4__ | PB4
__SW_D__ | Toggle channels | __PCINT5__ | PB5
__SW_E__ | Backlight | __PCINT6__ | PB6
__SW_PWR__ | Power | __PCINT0__ | PB0

[image]

charlespax, 2015-04-18:
Amazing!

closed 2015-04-18 by charlespax