PaxInstruments/t400-electronics #145: MicroSD card footprint

charlespax, 2015-02-01:
The MicroSD card connector does not match the component we have been prototyping with. We should make a new footprint that matches the component we will use in production.

The footprint should also indicate the centerline of the MicroSD card slot on the Measures layer.

charlespax, 2015-02-09:
See also https://github.com/PaxInstruments/t400-electronics/issues/79. Make sure the card positions are correct.

samchoy88, 2015-03-14:
According to the T-flash connect spec. I found Seedstuido library has slightly different on the pad.
[image]

Below picture is the T-flash connector in Seedstuido library.
[image]

The pin 4 should align to the line in below picture. And I updated component in Pax instrument library.
[image]

charlespax, 2015-03-18:
This change made it into the most recent board order, right? If so, we can close this and open another issue if it is still a problem.

closed 2015-03-18 by charlespax