PaxInstruments/t400-electronics #147: Charging status unstable while No battery installed

samchoy88, 2015-02-06:
MCU detected charging status unstable while no battery installed.
I check on the BATT_STAT test point seem stable at 2.5V but 3.4V at STAT pin of the charger IC.
May be the the voltage level is in marginal level and cause unstable.

samchoy88, 2015-02-06:
Changed the resistor from 10K to 100K and it seem stable now.
[image]

charlespax, 2015-02-06:
Based on a forum post I read (http://www.microchip.com/forums/m670010.aspx), we may need to add some capacitance on the battery terminals near the MCP73831. "I added another 4,7 µF ceramic capacitor, closer to the MCP73831 (right on the pins actually) and that seems to work too. So I guess it was some instability between the battery charger and the LDO."
[image]

samchoy88, 2015-02-06:
I found the battery disconnected made the charger IC not in Hi-Z status. It switch between Hi-Z and GND.

samchoy88, 2015-02-06:
This issue is different from forum post (http://www.microchip.com/forums/m670010.aspx).
As it has problem with battery charger IC supply to LDO directly and causing unstable voltage supply to circuit.
But my problem is the STAT pin give unstable signal when No battery connected.

samchoy88, 2015-02-06:
Fixed problem by this circuit. adding low pass filter and pull up resistor.
The 2k resistor and 10uF act as a low pass filter to avoid unstable signal from charging IC.
As I/O pin D5 do not have internal pull up resistor when act as an INPUT pin. A 10K resistor added externally.
[image]

charlespax, 2015-02-06:
Is there anything related to VBAT, VBAT_SENSE, or VBAT_EN floating that is causing this?

[image]

[image]

charlespax, 2015-02-06:
Okay, cool. When we're connected to USB the current drawn when STAT is low doesn't matter. When we are not connected to USB power STAT is Hi-Z, so the current draw should be negligible. Nice.

Can we close this issue?

samchoy88, 2015-02-06:
Problem fixed.

closed 2015-02-06 by samchoy88