PaxInstruments/t400-electronics #148: LCD quality issue

samchoy88, 2015-02-06:
One of the LCD sample has broken line.
I think this is quality issue of LCD factory.
[image]

charlespax, 2015-02-06:
Does this LCD have the 0.5 mm pitch cable or the 1.0 mm pitch cable?

Was the LCD this way the first time you turned it on or did it happen later?

samchoy88, 2015-02-06:
This is 1mm pitch cable. This happen after few turn on.

charlespax, 2015-02-06:
Okay, I just want to be clear before I talk to the manufacturer. The first time you turned on the LCD it had no problems, correct? Then you turned it off and back on a few times and it was still okay, correct? Then you turned it on again and the LCD was bad, correct?

charlespax, 2015-04-29:
I haven't seen this again. I'm moving this to the production milestone, so we can make indicate the PCBA factory should check for this.

Tsillen, 2015-10-06:
this is probably the zebra connector