PaxInstruments/t400-electronics #150: SPI bus design

charlespax, 2015-02-09:
I'm posting this application note about SPI bus design. I don't think we need this for the T400 thought. Moving to Ice box. http://hackaday.com/2014/11/25/better-spi-bus-design/

samchoy88, 2015-03-14:
It looks great and better design to avoid unexpected failure on SPI bus.

samchoy88, 2015-03-14:
[image]

Here are the reference reason of this design.
1. Add Pullup Resistors for Chip Select & Reset Signals: 
Without a pullup resistor, the second device can "hear" and respond to the communication taking    place on the first device, if that second device's chip select pin is not pulled up.

2. Verify tri-state behavior on MISO: use a tri-state buffer chip if necessary.
Most SPI chips will tri-state (effectively disconnect) their MISO pin when their chip select signal is high (inactive). However, some chips do not have proper MISO tri-state behavior.

3. Protect bus access with SPI.beginTransaction(settings) and SPI.endTransaction().
Newer versions of Arduino's SPI library support transactions.  Transactions give you 2 benefits:
Your SPI settings are used, even if other devices use different settings
Your device gains exclusive use of the SPI bus.  Others will not disturb you.

charlespax, 2015-08-08:
Will not do. Closing issue.

closed 2015-08-08 by charlespax