PaxInstruments/t400-electronics #152: Improve power on/off design

cibomahto, 2015-02-12:
The V.10 board I received seems to have this issue, not sure if it was present beforehand-

The 3.3V line hovers at 1.7V when it should be in an off state (3V3EN is low). It appears there is voltage leaking through somewhere. I tried the following:

Cut SW_PWR connection to 32U4 (near the 32u4)
Cut PWR_ONOFF connection to 32U4 (near the 32u4)
Cut VBUS connection to 32U4 (near the 32u4)
Pull R20
Pull R47
Cut 3.3V line after regulator

With the above changes made, the 3.3V line stays at 0V.

The well-labeled, accessible test points were super useful in debugging this.

Afterwards, I replaced VBUS and re-connected the 3.3V line, and the voltage output is still clean (0V). I re-installed SW_PWR and PWR_ONOFF, and the issue returned- 3.3V hovered at a high voltage again.

I tried creating a voltage divider (replace R6 and R7 with 10K resistors, place a 100k resistor between SW_PWR and the 32u4, and PWR_ONOFF and the 32u4), but this didn't help- the voltage on the 3.3V line is still .7V. when it should be 0V.

I think the problem is leakage through the processor I/O pins, and the solution is that they can't be connected directly to those connections as they currently are (this is probably an issue for VBAT_SENSE, VBAT_STAT and VBAT_EN as well)

One solution, for the power circuit, is to go back to the OR gate design. I'd prefer to see a latched design, but the OR one should still work (I'll check a V.9 board tomorrow).

charlespax, 2015-02-14:
I'm in my phone right now, so it is difficult to dig into this, but here are my thoughts.

- Which resistors are R6, R7, R20, and R47? I can only view the PDF and they aren't labeled.
- would it make sense that we have current leaking though the internal pull up resistors? We have VBUS connected to PWR_ONOFF and SW_PWR by the 100k pullup resistors, which are then connected to 3.3V inside the MCU by the internal pull up resistors on those lines (I think).
- If that is the problem, can we disable the internal pull up resistors or use pins that maybe do not have pull up resistors?
- Is this problem happening when the unit is off and not plugged into USB?

charlespax, 2015-02-14:
When the battery is connected and USB is not connected VBAT_SENSE and VBAT_EN are also connected to VBAT/VBUS through the 5k and 10k resistors. If SW_PWR and PWR_ONOFF are causing problems, VBAT_SENSE and VBAT_EN should be causing problems too, right?

charlespax, 2015-02-14:
I reread you comment and see you already mentioned those pins.

I suppose having 3.3V hover above 0V is a problem only if it is draining the battery in a meaningful way.  Maybe we should tap into VBUS and see if any current is being drawn.

charlespax, 2015-02-14:
If we leave things as they are, what is our power drain on the battery? It only matters if it matters.

charlespax, 2015-02-14:
We could potentially run the MCU 3.3V through a MOSFET and cut off current from draining to the 3.3V line. I wonder how much, if any, current would then drain through the MCU to ground.

charlespax, 2015-02-14:
I don't know what the solution is or if we can just leave things as they are, but we should address this in the next board revision. Moving to Electronics version 0.11.

cibomahto, 2015-02-15:
I should be more clear- this prevents the processor from turning off the regulator.

cibomahto, 2015-02-15:
It's not a good design to pull pins higher than Vbus (and Vbus is 0V at that time).

I'm pretty sure the previous circuit worked fine, so I would suggest re-testing a V.9 board to see the power on/off circuits work correctly. I blew up the 3.3v regulator on my v.10 board during this testing and stole it from the v.9 board, so I unfortunately don't have the parts around to retest that :-/.

charlespax, 2015-02-16:
@samchoy88 this sounds like a pretty serious issue. What are you thoughts?

samchoy88, 2015-02-16:
I discuss the issue with Matt, please don't mind the 3.3V regulator blew up in testing.
However, the circuit have many issue in most I/O pin connected to non-3.3V.
As the power may leak to/from I/O during sleep and operating.

Matt suggest to re-design / take special care of those pins.

Here are the list of I/O with issue or potential issue:
- BATT_STAT
- VBAT_SENSE
-VBAT_EN
- POR_ONOFF
- SW_PWR

samchoy88, 2015-02-16:
Here is the solution to improve battery measure circuit.
http://www.microbuilder.eu/Tutorials/Fundamentals/MeasuringBatteryVoltage.aspx

[image]

(Reference from microBuilder.eu)
This example extends on the basic voltage divider by allowing you to enable or disable the divider with a simple GPIO pin (BATTSENS_EN).
Q1 is a P-Channel MOSFET that will selectively allow the battery current to flow throw the voltage divider and into the ADC.  
Q2 is an NPN Transistor that will turn the P-Channel MOSFET (Q1) 'on' or 'off'.  The extra transistor is necessary because the battery voltage may well be above the maximum level that the GPIO pin can handle, which would almost certainly damage the MCU if they were connected directly.

By setting the GPIO line high, the transistor (Q2) will open the flow of current through the MOSFET (Q1), allowing you to safely measure the current battery voltage on the ADC line.  By setting the GPIO line low, you shut off the flow of current to the ADC and the voltage divider and can be certain that you are not needlessly using any of the battery's stored energy.

charlespax, 2015-02-16:
Are able to use the same part number as we already have in the design?
[image]

samchoy88, 2015-02-16:
The P-MOSFET can be replace by the DMP2035U-7.
Beside, the mosfet we are using has low leakage current.
[image]

samchoy88, 2015-03-04:
Added Mosfet to isolate the MCU (operate at 3.3V) and USB/Li-Po Battery (operate at 5V / 4.2V).

Updated Battery detect circuit
[image]

Updated Soft power circuit
[image]

charlespax, 2015-03-04:
Nice. Closing issue

closed 2015-03-04 by charlespax