PaxInstruments/t400-electronics #154: PCB locating feature

charlespax, 2015-02-19:
We have introduced a PCB locating feature into the enclosure to aid the assembly process and secure the PCB. See https://github.com/PaxInstruments/t400-enclosure/issues/80

The PCB needs corresponding slots milled into the sides. @starno can you give @samchoy88 some dimensions for this?

starno, 2015-02-27:
@samchoy88 the slots are symmetrical on the pcb.  the center on each slot is 15.28mm from the top edge of the PCB:
[image]
and each slot is 2mm long and 0.9mm wide: 
[image]
The end of each slot is round matching the milling bit that cuts it.  I'd assume a 0.9mm round there.

samchoy88, 2015-02-27:
Added slot on each side.
[image]

starno, 2015-02-28:
looks good

charlespax, 2015-03-01:
Awesome! Let's close this.

closed 2015-03-01 by charlespax