PaxInstruments/t400-electronics #155: QFN thermal pad

charlespax, 2015-02-21:
The ATmega32U4 QFN thermal pad has a single via to transfer heat from the MCU to the bottom side. Should we add more vias?

[image]

Tsillen, 2015-02-21:
no need

charlespax, 2015-02-21:
[citation needed]

Tsillen, 2015-02-21:
You could add up all the loads and calculate the temperature rise etc... but that is a lot of work.
The TQFP version does not have any over heating issues as shown by arduino using that format on their products.

I see that you have a few IO and some busses etc. The current draw of the chip will be (very)low. In that case no extra "cooling" is needed. 

But if you really want add a few more via's to it. It won't do you any good as it might make the reflow process work less good due to the increased thermal resistance that the big bottom pad will have.,

charlespax, 2015-03-15:
Cool. Let's leave things as they are. Moving to ice box.

charlespax, 2015-08-08:
Will not do. Closing issue.

closed 2015-08-08 by charlespax