PaxInstruments/t400-electronics #156: Internal analog reference voltage

charlespax, 2015-02-22:
Is this a problem?

Comment from @samchoy88 in https://github.com/PaxInstruments/t400-electronics/issues/43:
_As the Reference voltage is critical for calculating the VBAT voltage value. I expect using internal 2.56V voltage reference. and calculate the VBAT level by ADC x 2.56V / 1023. However, I found the VBAT value does not match with measuring point. After checking the circuit and ARef pin, the internal voltage reference only supply 2.508V and causing incorrect result in calculation._

charlespax, 2015-04-29:
@samchoy88 this has been implemented, correct?

MLXXXp, 2015-04-29:
The ATmega32U4 datasheet specifies that the internal reverence is typically 2.56V but can vary between 2.4V and 2.8V. See page 385:
http://www.atmel.com/images/doc7766.pdf

If you need a more precise reference value, for battery measurement and other purposes, you may have to measure and calibrate each individual unit by storing the measured reference voltage (as measured on the AREF pin) in the internal EEPROM, and use the stored value for calculations. If you do this, it would be nice to allow users to re-calibrate their unit by entering a new value, using a menu item, that they measured themselves.

The other option is to use a more precise external reference connected to the AREF pin.

charlespax, 2015-08-06:
@samchoy88 how should we address this?

MLXXXp, 2015-08-06:
I don't have Eagle installed, so I can't look at your latest schematic. If you're still using a Micrel MIC5219 to provide 3.3V then note that its output is specified to be within 1% (or 2% over the full temperature range). If this tolerance is sufficient for your purposes, you could tie this 3.3V output to the AREF pin for use as a reference.

If you need to be more precise, then as I've said, you could add a better external reference or allow for software calibration.

If it's just the battery voltage that you need to measure accurately (not other things relative to the reference) then another option is to add a potentiometer in the resistor divider that provides VBAT_SENSE, to allow hardware calibration. However, if you decide to calibrate each unit, I think software calibration would be preferred.

If you decide to use 3.3V or other higher external reference voltage, then you should also consider changing the values of the VBAT_SENSE resistor divider to take advantage of the larger range. With a 3.3V reference, using 18.7K on the high side and 34K on the low side will give a nice 5mV per step from the ADC.

samchoy88, 2015-08-12:
The battery voltage below 3.6V could be consider as low battery.
We can consider the battery level is always higher than 3.3V.
I will apply MLXXXp suggestion and use AVCC (regulated 3.3V) as ADC reference and using 18.7K on the high side and 34K on the low side will give a nice 5mV per step from the ADC.
Besides, a capacitor also added between ADC input and ground as noise filter.
See below update circuit:
[image]

MLXXXp, 2015-08-12:
A couple of notes:

1. The divider resistors should be 1% tolerance or better. Since 18.7 and 34.0 are EIA E96 (1%) values, whatever you buy probably will be, but unless all your resistors are specified as 1%, you should probably specifically mark these ones to be (at least) 1%, just to be safe.

2. The added 0.1uF capacitor will take some time to charge and settle after switching on Q2. Make sure you add a delay in your software to allow for this before starting the ADC conversion. Also make sure the ADC is enabled with the VBAT_SENSE input selected during this time, to allow the internal sample/hold capacitor to settle as well.

cibomahto, 2015-08-15:
Testing this, might want something a little higher than .1uF since there seems to be some noise spikes on the signal.

samchoy88, 2015-08-18:
For the resistors, we are using 1% value in general. Only the thermocouple part using 0.1% resistors.
(please see issue: Does the thermocouple filtration circuit affect the ADC readings? #53)

For the capacitor at VBAT_SENSE, I put 1uF as current design. Please turn on MOSFET with longer duration before taking ADC value.

samchoy88, 2015-08-18:
Schematic changed.
[image]

closed 2015-08-26 by charlespax

samchoy88, 2015-09-20:
18.7K and 34K resistors can be replace by value 11K and 20K resistors to achieve 50mV per ADC step with 3.3V ADC reference.

MLXXXp, 2015-09-20:
11K and 20K give exactly the same ratio as 18.7K and 34K, so either set will work pretty much identically.