PaxInstruments/t400-electronics #159: Battery charge controller

charlespax, 2015-03-19:
Maybe we should change the MCP73831 or how the STAT is connected.

[image]

cibomahto, 2015-03-19:
MCP73832 has a open drain output, maybe we should use that one isntead

cibomahto, 2015-03-19:
Otherewise there's a chance that VBUS_IN gets connected to STAT output through the MCP73831

charlespax, 2015-04-29:
@samchoy88 this has been addressed, right?

samchoy88, 2015-06-04:
Yes...this problem has been addressed....In Rev0.12 the board is using MCP73832T-2DCI/OT which is variance of MCP73831 with open drain output. But the circuit still keep Rev 0.12, as we can DNP or replace by 0 Ohm resistors.

closed 2015-06-17 by charlespax