PaxInstruments/t400-electronics #160: Bug in soft Power circuit

samchoy88, 2015-03-20:
As previous soft power circuit (using OR gate or NAND gate) required MCU running in application circuit and keep enable 3.3V regulator operate. But it required user press and hold Power button until MCU running application program, otherwise it system fail to start while user release button in bootloader program.
   
In order to turn on 3.3V regulator instantly, I changed the logic gate to SR Flip-Flop and using MOSFET to isolate the 5V and 3.3V. User can turn ON 3.3V regulator once they press the power button. SR Flip-Flop will hold the 3.3V regulator working until Flip-Flop received the high signal from MCU (inverses to Logic LOW by N-MOSFET for RST input at Flip-Flop) and turn SR Flip-Flop output LOW and off the regulator. (See below circuit)

Soft power circuit:
[image]

In practical, once user pressed power button and turn ON 3.3V regulator. Then the I/O (SW_PWR) that turn off Flip-Flop will raise voltage a little bit (around 60mV). This voltage level just meet Vgs and trigger MOSFET ON to give RST signal and turn OFF 3.3V regulator immediately.

That means the power will turn ON while I pressed the button, but it also turn off immediately when I release.

samchoy88, 2015-03-22:
Put a 10K pull low resistor at Q4 MOSFET gate can eliminate the bug.

charlespax, 2015-03-22:
It looks like the functions of SW_PWR and PWR_ONOFF are reversed. SW_PWR is

charlespax, 2015-03-22:
put signal to the MCU

samchoy88, 2015-04-03:
Bug fixed: 
1. Identified the short of JP1 cannot not hold 3.3V always ON. Updated R21 position.
2. fixed the bug in reversed signal name between SW_PWR and PWR_ONOFF
3. Added the pull down resistor for mosfet and prevent noise trigger during initial stage

Before:
[image]

After:
[image]

charlespax, 2015-04-04:
Do you think we can close this?

closed 2015-04-18 by charlespax