PaxInstruments/t400-electronics #161: MCP9800 part number

charlespax, 2015-03-23:
We should specify which MCP9800 version we are using. It should be MCP9800A0 which gives and i2c address of 000. If we don't specify this, we may end up getting MCP9800A5, which has an address of 101.

[image]
[image]
[image]

samchoy88, 2015-04-03:
Updated schematic.
Clearly mark down the part number in schematic and its relationship between I2C slave address.

[image]

charlespax, 2015-04-04:
Great. Can we close this?

closed 2015-04-18 by charlespax