PaxInstruments/t400-electronics #162: LCD connector

charlespax, 2015-03-26:
The current LCD connector is not very good for hand soldering. It would be nice to find a better one. We may alternatively change the footprint, giving a larger pad such that a soldering iron can transfer heat to the pad.

It would also be nice to have a flip top version rather than one that is pressed in parallel to the board.

charlespax, 2015-04-08:
We also discussed a flip top connector, but I think it's not available in this pin pitch.

If we change the footprint such that the pads are more exposed, we should be able to hand solder more easily. @samchoy88 did you change the footprint?

samchoy88, 2015-04-08:
I made change for Electronics version 0.12.
I took FH12-12S-1SH(55) FPC flip-lock connector as component reference.

[image]

Extended the pads at the side for hand soldering easily and adapt to both flip-lock or insert lock connector.
 
[image]

charlespax, 2015-04-09:
Awesome! I think this is going to make assembly easier, especially with that flash chip next to the LCD connector. Closing issue.

closed 2015-04-09 by charlespax