PaxInstruments/t400-electronics #163: Power draw

charlespax, 2015-04-06:
I believe the battery charging IC is set to charge at 500 mA. That doesn't leave anything for running the device if USB power is limited to 500 mA on the host side. Maybe we should change the configuration resistor such that the battery charges at 480 mA.

If I have a battery with a very low charge installed in the T400 and the T400 is powered on, the device will turn off when the USB is inserted into power. I believer the battery charger is taking all available current from the USB power and leaving none for running the MCU. After changing for a few moments plugging into USB does not power off the T400.

This is probably a problem for any charge level less than what will operate the T400. I suggest we reduce the battery charging rate by an appropriate amount.

samchoy88, 2015-04-08:
According to the datasheet MC73831, the current programming resistor should be 2.08K (round up to 2.1K) Ohms for 480mA charging current limit.
[image]

charlespax, 2015-04-08:
If we use 2.1k 1% resistors, the charge current can range from 473.9 mA to 478.5 mA. Using 5% resistors might be okay too. Using 10% resistors would put us in the danger zone. Please specify 1% resistors.

1%
1000/(2.1 + 0.01) = 473.9336492891
1000/(2.1 - 0.01) = 478.4688995215

5%
1000/(2.1 + 0.05) = 465.1162790698
1000/(2.1 - 0.05) = 487.8048780488

10%
1000/(2.1 + 0.1) = 454.5454545455
1000/(2.1 - 0.1) = 500

samchoy88, 2015-04-08:
More information will be provided in schematic.
[image]

MLXXXp, 2015-04-08:
Charles,
Your calculations are wrong. You don't just add 0.01 to get 1% of 2.1K. Likewise for 5% and 10%

1%
1000/(2.1 + 0.021) = 471.475719
1000/(2.1 - 0.021) = 481.000481

5%
1000/(2.1 + 0.105) = 453.514739229
1000/(2.1 - 0.105) = 501.253132832

10%
1000/(2.1 + 0.21) = 432.9004329
1000/(2.1 - 0.21) = 529.100529101

MLXXXp, 2015-04-08:
There is also a fairly large tolerance for charge current in the MCP73831 itself. See the *Current Regulation (Fast Charge Constant-Current)* section on page 3 of the datasheet.

Plus, there's some variation of charge current based on supply voltage and ambient temperature. See *FIGURE 2.6* on page 7 and *FIGURE 2.7* on page 8 of the datasheet.

http://ww1.microchip.com/downloads/en/DeviceDoc/20001984g.pdf

charlespax, 2015-04-09:
Oh crap! I meant to multiply 2.1 by those numbers. I feel really foolish right now :-(

Anyway, what should be the value of the PROG resistor to ensure our charge rate doesn't exceed 480 mA?

MLXXXp, 2015-04-09:
From what I can tell, the range of values given in the *Current Regulation (Fast Charge Constant-Current Mode)* section of the datasheet already takes into account voltage variations over the desired operating range, so we don't have to additionally refer to the graph in *FIGURE 2.6*.

Basically, it seems that the actual charge current can vary by +/- 10% of the value calculated. This is for ambient temperatures from -5°C to +55°C, but *FIGURE 2.8* indicates that charge current only goes lower outside this range, so it's not of concern.

So, to guarantee that the charge current doesn't exceed 480mA, we have to take 480mA as being 10% higher than the typical charge current. 480mA / 1.1 = 436mA.

Using the formula for the PROG resistor: 1000 / 436 = 2.2936

Then there's the tolerance of the resistor to take into account. Since the chip itself can vary by 10%, I'd specify a 1% tolerance resistor to minimize any further variance. So, 2.2936 would be the low side of the 1% tolerance range. 2.2936 / 0.99 = 2.3168

The closest EIA 1% value would be 2.32K

With a 2.32K resistor the typical charge current would be 1000 / 2.32 = 431mA

The highest and lowest possible charge currents, using 10% for the chip and 1% for the resistor would be:
Maximum 479mA
Minimum 384mA

charlespax, 2015-04-18:
Thanks, @MLXXXp nice work. @samchoy88 should we go with the 2.32K resistor then? What are your thoughts?

samchoy88, 2015-04-20:
I agree with @MLXXXp calculation, I have no problem to change the resistor to 2.32K.
For my understanding, most USB power supply can support 10% tolerance. It means the supplier can supply up to 550mA. Besides, the power consumption of the system is not significant compared with charging current. But it help to avoid power supply shut down when charging.

charlespax, 2015-04-20:
Great. Let's do it.

samchoy88, 2015-04-20:
Updated.
[image]

closed 2015-04-20 by charlespax