PaxInstruments/t400-electronics #166: Battery reverse polarity protection

charlespax, 2015-04-08:
I would like to know if the T400 power system is protected against connecting a battery with reverse polarity. This is relevant to a situation where a user is hacking the device to run off an external power source via the battery connector or if they operate the device without an enclosure and use the wrong battery.

If USB power is connected, what happens if a battery is connected with reverse polarity?

If USB power is disconnected, what happens if a battery is connected with reverse polarity?

samchoy88, 2015-04-09:
If USB power is connected, what happens if a battery is connected with reverse polarity?
If USB power is disconnected, what happens if a battery is connected with reverse polarity?

Both of the above case will result to short circuit. But short circuit only happen in MCP73831 battery charger. As MOSFET Q1 do not turn on and it becomes the barrier of the reverse current.

[image]

samchoy88, 2015-04-09:
I can add a P-MOSFET to prevent the battery polarity reversed. But it still contribute a little bit power lost and cost in circuit.
http://www.ti.com/lit/an/slva139/slva139.pdf

Besides, there is a protection circuit inside BL-5C battery.
See issue: Does the BL-5C have internal protection? #46

charlespax, 2015-04-09:
We can keep things as they are as long as nothing is going to blow up.

charlespax, 2015-04-29:
Closing issue.

closed 2015-04-29 by charlespax