PaxInstruments/t400-electronics #168: Update material list

samchoy88, 2015-04-20:
Update the attribute of each component on schematic except those resistors and capacitors.
Remaining 2 components has unknow part number or manufacturer.
- Unknow manufacturer of LG132643-DW
- Unknow manufacturer and Part no. of BL-5C battery connector

charlespax, 2015-04-24:
The LCD is a custom manufactured part based on the LG132643-DW. You can list the manufacturer as Pax Instruments and the part number of PI132643FFDWHUV-P1

Did you get the battery connector part number from the guy in the HQB booth?

samchoy88, 2015-04-24:
They only gave me the name of the connect in Chinese - "77电池座". And it is really common in market with the part number "BC-4-3P" for 3.0MM pitch 7.7MM 3P seat height of 77 Battery Battery Connector.
So, I will mark it in description as "Right angle battery connector, 3P, Pitch=3mm, 7.7mm height"
part number is "BC-4-3P"

charlespax, 2015-08-06:
Is the material list finished? Can we close this?

samchoy88, 2015-08-10:
Yes, all critical parts (except the common parts i.e resistors and capacitors) must have the following attribute:
1. MF (Manufacturer / Supplier / Brand)
2. MPN (Manufacturer Part number)
Additional note may provide in schematic. see below example:
[image]

These attributes also exist in BOM list
[image]

charlespax, 2015-08-11:
Done. Closing issue

closed 2015-08-11 by charlespax