PaxInstruments/t400-electronics #169: MCU VBAT_SENSE on PD6 is not Analog Pin A10

samchoy88, 2015-04-26:
I got incorrect analog reading from analogRead(VBAT_SENSE);
I finally found the MCU component in schmatic created by sparkfun named the pin PD6 incorrectly.
someone posted the issue on Arduino forum and find the mistake too.
http://forum.arduino.cc/index.php?topic=82992.0
[image]

[image]

samchoy88, 2015-04-26:
Just required to correct the schematic of PD6 pin become D12/A11

charlespax, 2015-04-27:
Did you change this in our part library and update the schematic and silkscreen?

samchoy88, 2015-06-04:
Not yet change on schematic and silkscreen, please confirm to change

charlespax, 2015-06-17:
Yes, please make our parts correct.

samchoy88, 2015-08-10:
According to the pin assignment in pins_arduino.h.
[image]

the part library corrected as follow:
Before:
[image]

After:
[image]

closed 2015-08-11 by charlespax