PaxInstruments/t400-electronics #170: Thermocouple connector footprint

charlespax, 2015-04-29:
We need a footprint for the thermocouple connector (https://github.com/PaxInstruments/t400-electronics/issues/17).

The connector should overhang the PCB a bit. @starno can you give dimensions for this? The PCB will need a slot for the vertical tab and a cutout for the 30 degree angled tab on the back.

This should go on the next revision.

Here is the part drawing.
[image]

MLXXXp, 2015-04-29:
Should the cutout for the 30 degree angled tab be plated through and around, and exposed on the solder mask, to allow the tab to be soldered? This could give more mechanical strength to the clip to board bond.

charlespax, 2015-04-29:
Here is a bottom view of the connector.
[image]

I imagine the footprint in the image below. The vertical tab would be a milled slot that is plated through. The two slots for the 30 degree tabs would be milled and unplated. The area under the connector would be a solder pad, some of which would stick out a little from the connector to aid hand soldering.
[image]

I don't think there would be any harm in plating the thirty degree slots, but I don't think they would be soldered in the SMD process. I think I would want that area clear of solder, so it doesn't interfere with the male connector. I think if someone hand soldered the thirty degree slot area from the bottom, it would be fine. Not really sure what's best.

starno, 2015-05-04:
Ok, with the latest rev connector, a **1.5mm overhang** off the top edge of the PCB will give ~0.5mm clearance between the top panel and connector inlet.  
[image]
[image]

Lets go with 1.5mm off the board.  (distance does not include the bent inlet portion)

charlespax, 2015-05-04:
Here is a technical drawing. There should be enough information to make a footprint.
[image]

samchoy88, 2015-06-04:
What is the part number of this clip?

charlespax, 2015-06-17:
Manufacturer: Pax Instruments
Product number: 1024
Link: http://paxinstruments.com/products/sku1024/

samchoy88, 2015-06-18:
1024 TC connector is ready in Pax instrument library
See https://github.com/PaxInstruments/PaxInstruments-Eagle-Libraries
Footprint locate In the PaxInstruments-Connectors.lbr

[image]

Preview Footprint on T400 PCB.
[image]

charlespax, 2015-06-18:
When I was on my phone I thought you used the old design files in the electronics repository. Now that I look at the first post of this issue I see the correct drawing. I think everything should be good then. I'll order these boards right away. Thanks, Sam.

closed 2015-06-18 by charlespax