PaxInstruments/t400-electronics #172: Fuse temperature

charlespax, 2015-04-30:
The fuse gets pretty hot.

Revision 0.12 of the PCB was installed in the enclosure with a mostly discharged battery. After allowing the battery to charge for about thirty minutes I opened the enclosure and took several thermal images. The highest recorded temperature of the PTC fuse was 151 C.
[image]

After letting the case sit open for a few minutes the temperature decreased to 132 C
[image]

During this time the ADC was approximately 86 C. We should check to see how this temperature can affect the ADC readings.
[image]

Even with the high temperature of the PTC the temperature across the isothermal plane was at a consistent 83.3 C. I believe this demonstrates the effectiveness of the thermal isolation slots and the thermal conductivity of the isothermal plane.
[image]
[image]
[image]

samchoy88, 2015-06-04:
We are using MF-MSMF050 fuse in this circuit.
This is the specification of the fuse
http://www.mouser.com/ds/2/54/mfmsmf-72300.pdf
[image]

It seem the resistance is relatively low 0.15Ohms ~ 1 Ohms.
When it is charging, it raise temperature to 151 Fahrenheit (66 Celcius).
If it is necessary to reduce the internal heat source, I recommend not to use fuse.
As all  PTC Resettable Fuses using similar technology to cut off circuit when temperature raise to certain level.

Besides, the Time to Trip is relatively long (around 1 second @ 2.5A current). Some of the components may damaged within that period.
That means the fuse protection may not effective.

[image]

charlespax, 2015-06-17:
Is it safe to remove the Fuse? If so, we should remove it.

samchoy88, 2015-08-05:
I think we should remove it.

charlespax, 2015-08-06:
Okay, let's remove it.

samchoy88, 2015-08-12:
I think this case can be close after removed the fuse

samchoy88, 2015-08-25:
Fuse removed in Rev 0.13
[image]

closed 2015-08-25 by charlespax