PaxInstruments/t400-electronics #178: RTC battery

charlespax, 2015-06-03:
We need to select a battery part number for the RTC battery. We need to confirm it works with the electronics.

samchoy88, 2015-08-09:
The following brand are the reliable brand/suppliers in Asia region.

Brand: GPI Int'l Ltd
http://www.gpbatteries.it/allegati/datasheet/specialistiche/litio/103123_CR1620.pdf

Brand: Maxell
http://www.maxell.com.tw/images/uploads/2014/10/CR1620_DataSheet_e.pdf

Brand: Panasonic
http://industrial.panasonic.com/lecs/www-data/pdf2/AAA4000/AAA4000CE45.pdf

charlespax, 2015-08-09:
Please order samples of the batteries you think we should use.

samchoy88, 2015-08-12:
Purchased some 1220 batteries and test and Currently testing below brand
Brand: Sony
Part number: CR1220
http://www.sony.net/Products/MicroBattery/cr/pdf/cr1220.pdf

Brand: Maxell
http://biz.maxell.com/files_etc/9/cr/en/CR1220_DataSheet_e.pdf

charlespax, 2015-08-26:
Please add a note to the part that it takes a CR1220 battery.

samchoy88, 2015-08-27:
Added note in schematic
[image]

closed 2015-08-27 by charlespax