PaxInstruments/t400-electronics #179: Thermocouple connector placement

charlespax, 2015-08-08:
The thermocouple connectors must move closer to the top panel (https://github.com/PaxInstruments/t400-enclosure/issues/72). We could probably have the connectors hang over the PCB, but it would be best to extend the PCB top edge a little.

samchoy88, 2015-08-09:
How much I can extend at the upper edge?

charlespax, 2015-08-09:
You cannot extend the entire upper edge. About 70 mm to 74 mm of the top edge can be extended because it will have to fit inside the top panel recess as seen in the image below.
[image]

You can extend it between 2.5 to 2.75 mm as seen in this image.
[image]

samchoy88, 2015-08-10:
Thermo-clip hang out a little bit (below 2.7mm), but I worry the clip and conflict with the LCD stand at both left and right side.
[image]

charlespax, 2015-08-11:
I will continue work on the connector footprint. I've looked at the spacing using the physical LCD, PCB, and connectors I have, so I'm 100% confident that we can make a footprint that works. I will make another test PCB that includes several versions of the this area of the T400 board, each with a different footprint. Let's hold off on ordering 0.13 PCBs until I can test the new thermocouple connectors on the new test PCB. The factory says we'll have new samples tomorrow. I'll make the test PCB today and order it for rush delivery. I should have it Thursday or Friday.

charlespax, 2015-08-11:
I've updated the TC connector in the Eagle Library repo to tighten up the Dimensions. I checked it on the T400 PCB, but did not commit the changes. If this footprint works, it would look like this:
[image]

I've added another footprint test PCB to the repo to get the TC contacts as close as possible. We'll see which is best once we have the connectors this week and the test PCBs come in.
[image]

Don't worry about changing the TC connector on the T400 PCB right now. I'll let you know the results of the new test PCB. In the mean time you can continue work on the T400. Any changes to the footprint will be small and only have trivial, if any, affects on the layout.

samchoy88, 2015-08-12:
Look better now.
[image]

closed 2015-08-26 by charlespax