PaxInstruments/t400-electronics #184: Move SW_PWR off of PB0/SS

cibomahto, 2015-08-13:
SW_PWR unfortunately needs to be moved off of PB0(SS), since the SS pin cannot be used as an input while the hardware SPI port is enabled. It's actually best to avoid PB0 completely, since it is toggled uncontrollably when the SPI port is initialized. I was able to get the power circuit working correctly by making these changes:

Move SW_POWER to PB7 (D11)
Move POWER_ONOFF to PD5 (TXLED/D30)

samchoy88, 2015-08-18:
According to current design, PB7 assigned to switch OFF system by I/O pin (PWR_ONOFF).
If we avoid using PB0/SS pin, is it OK to change using PD5 as the control pin?
I saw Matt released issue (Procedure for modding V12 board to act like a V13 board #190).
And give instruction on "Disonnect PWR_ONOFF from PB7 and connect to PD5", but it already assign as TXLED in Arduino.

[image]

Potential problem may raise after this change, I hope this issue confirmed once again in here.

samchoy88, 2015-08-18:
Since we still have 2 reserved I/O pin.
We can use A0 or A1 pin as PWR_ONOFF output pin and control device OFF.
This avoid using TXLED (Pin 22 <PD5>) and RXLED (Pin 8 <PB0>).

[image]

cibomahto, 2015-08-18:
Please follow my suggestion exactly. I've already modified the board design and bootloader in arduino so that TXLED can be used as D30.

closed 2015-08-26 by charlespax