PaxInstruments/t400-electronics #185: Clock frequency

charlespax, 2015-08-13:
The firmware could use some extra clock cycles to get things done. We may be able to kick things up to 12 MHz on our 3.3V or even overclock at 16 MHz.

charlespax, 2015-08-13:
The datasheet gives us maximum frequency versus Vcc:

To calculate the max clock frequency we use:
m = (16 - 8)/(4.5 - 2.7)      // m = 4.44444
x=3.6 V
Fmax =m × (x-2.7) + 8      // Fmax = 12 MHz, at 3.3 V Fmax = 10.6666 MHz

charlespax, 2015-08-13:
Based on this information and my conversation with Matt, I want to change the clock to 12 MHz.

MLXXXp, 2015-08-13:
I'm pretty sure that you have to run at either 8MHz or 16MHz if you want to use full-speed USB.

See sections 6.1.5 and 21.4 in the ATmega32U4 datasheet.

charlespax, 2015-08-13:
Good call. I think I saw something about that in the datasheet. I'll give it another look and cite the information. Removing from milestone 13 for now.

cibomahto, 2015-08-13:
Good catch. Bummer that 12MHz isn't an option! It seems we need full-speed USB to do CDC-ACM.

charlespax, 2015-08-13:
Is there a place in the datasheet that explicitly states this? I wonder if the internal oscillator can be used for the USB prescaler.

charlespax, 2015-08-14:
For Full Speed operation the USB controller requires a 48 MHz signal from the internal PLL, which would require a signal from an external clock source for Full Speed. The internal RC can be used as input to the PLL only in Low Speed USB operation.

Looking at the PLL prescaler, PLL, and PLL postscaler, there just isn't a way to multiply/divide to get the 48 MHz to the USB. It looks like we're stuck with 8MHz

[image]
[image]

closed 2015-08-14 by charlespax