PaxInstruments/t400-electronics #186: Move RTC_INT to PD3(INT3) and SW_B to PE6(INT6)

cibomahto, 2015-08-13:
INT6 is a special interrupt that needs to be configured as level sensitive in order work as a wakeup source for the processor. Unfortunately, the 1Hz reference signal from the RTC is a square wave with 50% duty cycle, so connecting it to INT6 and enabling the level sensitive interrupt means that the processor is stuck servicing this interrupt for 50% of the time (the alternative is to turn off level sensitivity and never sleep the processor)

I’d like to move RTC_INT off on INT6 and onto INT3, to mitigate the weirdness of INT6 as much as possible. This will push the issue to SW_B, where we can mitigate it with a workaround (disable the INT6 source as soon as it is triggered, and only re-enable it when going back to sleep).

Combining this and #184, the new interrupt table is:

PB0: PCINT0: (can’t use, SS)
PB1: PCINT1: (can’t use, SCK)
PB2: PCINT2: (can’t use, MOSI)
PB3: PCINT3: (can’t use, MISO)
PB4: PCINT4: SW_C
PB5: PCINT5: SW_D
PB6: PCINT6: SW_E
PB7: PCINT7: SW_PWR
PD0: INT0: (can’t use, SCL)
PD1: INT1: (can’t use, SDA)
PD2: INT2: RTC_INT
PD3: INT3: SW_B 
PE6: INT6: SW_A

I’ll mod my Rev12 board tomorrow to verify that these changes will work.

cibomahto, 2015-08-13:
Here's a logic capture of what's happening:

[image]

cibomahto, 2015-08-13:
And note that we can't disable the INT6 interrupt to get around this, because it would prevent RTC_INT from being able to wake the processor up again later (or we'd have to keep the processor awake 50% of the time, while the 1Hz signal is asserted).

The user isn't likely to hold SW_B down for long periods of time, so putting the disable-then-reeenable-before-sleep workaround should work fine.

samchoy88, 2015-08-18:
According to the suggested Pin assignment

PB4: PCINT4: SW_C
PB5: PCINT5: SW_D
PB6: PCINT6: SW_E
PB7: PCINT7: SW_PWR
PD2: INT2: RTC_INT
PD3: INT3: SW_B 
PE6: INT6: SW_A

[image]

the PCB layout would be like this.
[image]

Is that possible to have some flexible on pin assignment?
PB4: PCINT4: RTC_INT
PB5: PCINT5: SW_D
PB6: PCINT6: SW_E
PB7: PCINT7: SW_C
PD2: INT2: SW_B 
PD3: INT3: SW_A
PE6: INT6: SW_PWR

then the PCB layout may be much better
[image]

cibomahto, 2015-08-18:
Two issues:
1. RTC_INT must be on INT2 or INT3 so that it has it's own interrupt vector. all of the PCINT interrupts share a single vector.
2. Understand that this will be a couple more hours of software work to change and verify the button interrupts.

samchoy88, 2015-08-18:
OK, that make sense.

charlespax, 2015-09-28:
This did not get done int version 0.13  :-(

charlespax, 2015-10-07:
The firmware and electronics match and are connected properly.

closed 2015-10-07 by charlespax