PaxInstruments/t400-electronics #187: AT90USB microcontroller

charlespax, 2015-08-14:
The AT90USB series should be considered for the next hardware generation.

The [Industruino](https://industruino.com/page/home) has an ATmega32U4 and AT90USB1286 version for each of their products. They have have [AT90USB1286 board definition files](https://github.com/Industruino/Board-definition-files) for the Arduino IDE ready to go.

## Operating frequency
The AT90USB has the same operating clock limitations as the ATmega32U4 (https://github.com/PaxInstruments/t400-electronics/issues/185). Running it at 3.3 V we would likely have to operate at 8 MHz to stay within specification and also use Full Speed USB.
[image]

cibomahto, 2015-08-14:
For comparison, the Freescale ARM chips are <$2.5 in quantity 500 (lowest i've paid was $1.40), and can run at 48MHz in those conditions, have 16k of ram, and 32-128k flash.

charlespax, 2015-08-14:
Okay, I changed "next hardware generation" to "T400-based development kit". The next generation will certainly be ARM. However, the ATmega32U4 and AT90USB218 appear to be nearly identical.

At a minimum we are releasing a dev kit based on the ATmega32U4. Hopefully, we can drop in the AT90USB128 with only trivial changes, so we can get something out the door quickly. Otherwise, I'd just go with the 32u4. I want to get people hacking with minimal effort.

charlespax, 2015-12-18:
Here is the AT90USB pin out with the ATmega32U4 pins noted.
[image]