PaxInstruments/t400-electronics #189: Suggestion: depopulate or remove external pullups on the buttons

cibomahto, 2015-08-15:
External pullups are usually only necessary when the traces from button to processor input are long, or expected to be subject to large amounts of external noise. Neither of these seems to apply here, and the internal resistors should work fine for the buttons.

[image]

samchoy88, 2015-08-17:
I think these resistors keep in PCB layout but DNP in material list.

charlespax, 2015-08-17:
Yes, Sam, let's keep them in the design as DNP. It can be useful to have them there in other derivative designs. For example, to make the LCD shield I just copied the Eagle files and removed a bunch of parts. It would be good to now have to add them back later. However, if we run into an issue we can solve by removing them, we should remove them. I made a note for production https://github.com/PaxInstruments/t400-electronics/issues/191

closed 2015-08-17 by charlespax