PaxInstruments/t400-electronics #190: Procedure for modding V12 board to act like a V13 board

cibomahto, 2015-08-15:
This is required to test the latest firmware:
1. Add 4.7k resistor between VUSB and ground (#188)
2. Disconnect SW_B from PD3(INT3) and connect RTC_INT in its place (#186)
3. Disconnect RTC_INT from PE6(INT6) and connect SW_B in its place (#186)
4. Disonnect PWR_ONOFF from PB7 and connect to PD5 (#184)
5. Disconnect SW_PWR from PB0(SS) and connect to PB7 (#184)
6. Connect VUSB to net VBUS_IN (#183)
7. Not shown: Change R20 to 18.7K and R24 to 34K (#156)
8. Not shown: Add capacitor between VBAT_SENSE and ground (#156)
9. Optional: Remove R16, R15, R8, R11, R12 (pull-ups are enabled in firmware) (#189)

[image]
[image]

cibomahto, 2015-08-16:
Original images were messed up
[image]
[image]

charlespax, 2015-10-03:
In step 2, SW_B is actually connected to PD2(INT2). This is in the schematic, the picture you uploaded, and the datasheet.
[image]
[image]

To correct step 2, 2. RTC_INT will be connected to PD2(INT2) and SW_B will be connected to PE6(INT6).

closed 2015-10-03 by charlespax

cibomahto, 2015-10-09:
Ok, that sounds correct.