PaxInstruments/t400-electronics #211: MicroSD card heating

charlespax, 2015-10-06:
I have some eekoo 1 GB cards I've been testing. So far two of them have gotten very hot. T400 issue or crappy MicroSD cards?

Tsillen, 2015-10-06:
There is no reason why an SD card should get hot. I would advice testing with some decent quality ones like Sandisk or Kingston. 

Worst case you might want to check which voltage you are providing for the SD cards. Maybe it is to high or even oscillation.

charlespax, 2015-10-06:
The other ones I have are fine. It's only been the eekoo ones. But I'll also look into this further to ensure it's not the circuit. Don't worry though, we'll ship with a reputable brand :-)

Tsillen, 2015-10-06:
Good. If you have questions about whatever just ask me on skype: tsillen

charlespax, 2015-12-16:
Shitty MicroSD cards. This has not appeared again.

closed 2015-12-16 by charlespax