PaxInstruments/t400-electronics #219: MCP73831 Note

charlespax, 2016-05-19:
Make sure the MCP73831 is specified in the schematic and BoM. Otherwise, there would be charge problems.

slowcoder, 2016-05-19:
That would probably be which version of the MCP73831 you intend to be used. The different MCP73831-X chips have different CC->CV thresholds (4.2V, 4.35V, etc)

samchoy88, 2016-12-10:
The exactly part number should be in Eagle schematic
The part number is MCP73832T-2DCI/OT
[image]

[image]

charlespax, 2016-12-11:
I've updated the library device and labwiz schematic to "MCP73831/2"

[image]

closed 2016-12-11 by charlespax